HKUST-Aerial-Robotics/VINS-Mobile #148: Wierd Display on iPhone XS Max

BigJohnn, 2024-03-03:
Hi, I have tested the code with my xs max, with small changes, 

https://github.com/BigJohnn/VINS-Mobile/tree/xsmax

everything is okay, except that the output image color is wierd...
I have saved the input/output cv::Mat( which seems okay) for some single draw call, and have recorded the test video:

https://github.com/BigJohnn/VINS-Mobile/tree/xsmax/ScreenCapture

Any idea? Thx!

GrooveWJH, 2024-12-03:
```
case iPhoneXSMax:
            printf("Device iPhoneXSMax plus param, 瞎写的。。\n");
            FOCUS_LENGTH_X = 526.600;
            FOCUS_LENGTH_Y = 526.678;
```

你这段都瞎写的，相机内参标定会影响一切。

BigJohnn, 2024-12-04:
> ```
> case iPhoneXSMax:
>             printf("Device iPhoneXSMax plus param, 瞎写的。。\n");
>             FOCUS_LENGTH_X = 526.600;
>             FOCUS_LENGTH_Y = 526.678;
> ```
> 
> 你这段都瞎写的，相机内参标定会影响一切。

还没到那一步，  是说渲染效果比较奇怪

closed 2024-12-12 by BigJohnn